sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #213: u-blox max-10m

hoka83, 2024-03-27:
does this library support U-BLOX MAX-10M please?

PaulZC, 2024-03-27:
Hi @hoka83 ,

For the MAX-M10, you will need to use [v3 of the library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3). Please see [this issue](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/issues/37) - the same is true for the MAX-M10.

Best wishes,
Paul

closed 2024-03-27 by PaulZC

hoka83, 2024-03-27:
but your mentioned issue is asking about MIA-M10 not MAX -M10, are these the same chips? @PaulZC

PaulZC, 2024-03-27:
"the same is true for the MAX-M10" - the MAX and MIA use the same interface protocol.

hoka83, 2024-03-27:
okay thanks to you Engineer Paul

hoka83, 2024-04-01:
Hello, How to active antenna supply by turnning on/off by the LNA_EN signal using the library?

hoka83, 2024-04-01:
Does also ublox MAX-M10 have an embedded antenna or i have to connect it with external antenna?

PaulZC, 2024-04-01:
Hi @hoka83 ,

Only the u-blox SAM modules have a built-in antenna. For the MAX-M10, you will need to add an external antenna.

For an active antenna, you need to add extra components. Please see the schematic for any of the SparkFun GNSS boards:

https://www.sparkfun.com/products/18037

https://cdn.sparkfun.com/assets/f/3/f/b/4/SparkFun_u-blox_GNSS_MAX-M10S_Schematic_v10.pdf

To use the LNA_EN pin, you need a more complex circuit. Please see the Integration Manual for more details:

https://cdn.sparkfun.com/assets/5/c/a/0/b/MAX-M10S_IntegrationManual_UBX-20053088.pdf

Best wishes,
Paul