sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #215: gnss modules and ATtiny chips

hoka83, 2024-04-04:
Are GNSS modules for example "UBLOX MAX-M10" familiar with ATtiny chips like ATtiny1616 ?

PaulZC, 2024-04-04:
Hello Mahmoud (@hoka83 ),

Are you asking if the u-blox MAX-M10 module will work with an ATtiny1616? Or are you asking if this library will run on ATtiny1616?

Best wishes,
Paul

hoka83, 2024-04-04:
yes paul, is it going to work?

PaulZC, 2024-04-04:
Are you asking if the u-blox MAX-M10 module will work with an ATtiny1616?
Or are you asking if this library will run on ATtiny1616?

hoka83, 2024-04-05:
if the library will run on ATtiny1616

PaulZC, 2024-04-05:
Hello Mahmoud (@hoka83 ),

There are a few parts to your question:

1. Does ATtiny1616 have an I2C interface, or a UART?
2. This is an Arduino library. Is there Arduino support - an Arduino Board package - for the ATtiny1616?
3. Does it have enough memory, both Flash and RAM?

Looking at the [Microchip product page](https://www.microchip.com/en-us/product/attiny1616):

It has I2C and USART. The USART can (probably) be configured as a standard UART.

Spence Konde (@SpenceKonde)'s [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) supports the ATtiny1616.

It has 16kB of Flash and 2kB of RAM. If I compile one of the library examples for the ATtiny1616 (with Optiboot), it needs approximately 30kB of program space and 1kB of RAM. I see the error ```region text overflowed by 17027 bytes```. So, no, it will not run on the ATtiny1616...

This is a big complex library. There are much simpler u-blox GNSS (UBX) libraries that may meet your needs.

Best wishes,
Paul

closed 2024-04-05 by PaulZC

hoka83, 2024-04-05:
Is it possible to run this library on ATtiny 1616 by adding external flash memory though?