sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #222: EXAMPLE12_USEUART - ZED F9P COMMUNICATION ISSUES WITH ARDUINO MEGA USING UART 1

Lucas0rt, 2024-05-22:
Hi everyone,

I am doing this post rearding the SparkFun_u-blox_GNSS_Arduino_Library example code "Example12_UseUart.ino", for reading lat and long via UBX binary commands using UART 38400 baud - free from I2C-> link:
[https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example12_UseUart/Example12_UseUart.ino]

I currently encounter issues while implementing this code. I'm using a ZED-F9P integrated on an Ardusimple SIMPLERTK2Budget board
[https://www.ardusimple.com/product/simplertk2b/] + Arduino Mega 2560 as CPU. The issue is that I can't establish the UART 1 connection between the two devices while using the Example12_UseUart.ino code.

**My wiring:**
Arduino 5V -> Vin F9P board
Arduino GND -> GND F9P board
Arduino IOREF -> IOREF F9P board -> mandatory for SIMPLERTK2Budget board -> **the board as its own level shifter converting the 3.3V signal of the ZED-F9P into 5V to communicate with the Arduino**
Arduino RX (PIN 19) -> TX F9P board (Serial1)
Arduino TX (PIN 18) -> RX F9P board (Serial1)
### Subject of the issue

**U-Center configuration:**
I did set-up in Message view -> CFG -> PORTS: Target UART1 / UBX messages as protocol in / UBX NMEA RTCM 3 messages as protocol out / baud rate 38400
Last version of firmware updated

**Arduino program 1 for UART communication: No issues**
While using this basic program with seria1.available(), I do well receiving NMEA messages and can read them in the Arduino serial monitor so it means the UART 1 communication is working.
```
void setup() {
  Serial.begin(9600);  // Communication IDE Arduino
  Serial1.begin(38400); // Serial1 communication with GNSS module
}
void loop() {
  while (Serial1.available()) {
    char c = Serial1.read()
    Serial.write(c); //Get NMEA messages on IDE
  }
  delay(3000);
}
```
**Arduino program 2 with Example12_UseUart.ino : The communication is not established**
Then, when I want to implement the Example12_UseUart.ino code, with few modifications so I use Serial1 with RX pin = 19 and TX pin = 18 on the Arduino Mega, the code gets stuck in the "Initialize librairy to use Serial1" step since it never switch to true.

```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h> //http://librarymanager/All#SparkFun_u-blox_GNSS
SFE_UBLOX_GNSS myGNSS;

void setup()
{
  delay(3000);
  Serial.begin(38400); //communication with IDE Arduino
  while (!Serial); // wait user open serial monitor
  Serial.println("SparkFun u-blox Example");

  // Initialize communication with GNSS module
  Serial1.begin(38400); 

  // Initialize librairy to use Serial1 
  if (myGNSS.begin(Serial1) == false)  //STUCK HERE -> never switch to true
  {
    Serial.println("u-blox GNSS not detected. Please check wiring.");
    while (1);
  }
  Serial.println("GNSS serial connected");
}
```

Does anyone would know why the connection is not made while using this code? Many thanks in advance.
Lucas

PaulZC, 2024-05-23:
Hi Lucas (@Lucas0rt ),

It looks like you are doing everything correctly. I don't know why the library is not beginning correctly.

Maybe the [ArduSimple configuration](https://www.ardusimple.com/configuration-files/) has something to do with it? I'd suggest using u-center and UBX-CFG-CFG to restore the default u-blox configuration. Select "Revert to default configuration" and click "Send". Then "Save current configuration" and "Send". You can restore their configuration again later if you want to by following the instructions in that link.

You can use UBX-MON-VER to confirm which firmware the module has installed. It should be 1.32 for best results.

A logic analyzer would help - to monitor both the TX and RX lines, and confirm the baud rate is correct.

I hope this helps,
Paul

PaulZC, 2024-05-23:
Your NMEA messages look OK. The prefix is GNRMC instead of GPRMC because the ZED-F9P is using multiple GNSS constellations. You do not need the ```delay(3000);``` in program 1. If you delete that, you should see messages arrive every second.

You can also use ```myGNSS.enableDebugging();``` _before_ the ```myGNSS.begin``` to enable additional debug messages. Please try that and copy and paste the output here.

Lucas0rt, 2024-05-23:
**Hey Paul! Wow, just thank you so much!**! It sounds easy now but it really made me crazy for days ^^
I indeed had installed configuration files from Ardusimple which led to this issue. I did reset the default configuration as you mentionned and it was because of this.

Also, in case some other people have this issue, I saw that if I put anything in the UBX -> CFG -> PRT in the protocol in, it is not working anymore and even if I put "none".
[image]

Here is the ouput in the IDE I get now (with in addition to the code the `myGNSS.enableDebugging();` before the `myGNSS.begin`):

```
19:40:06.600 -> Sending: CLS:NAV ID:PVT Len: 0x0 Payload:
19:40:06.600 -> sendCommand: Waiting for No ACK response
19:40:07.484 -> Incoming: Size: 92 Received: CLS:NAV ID:PVT Len: 0x5C Payload: C0 54 2D 18 E8 7 5 17 10 28 6 37 20 0 0 0 1F 7C F9 FF 3 1 EB 14 56 84 93 F 76 CF 7F 1A BD C8 1 0 BA 40 1 0 A B 0 0 BB 17 0 0 5 0 0 0 F9 FF FF FF FC FF FF FF 9 0 0 0 40 8 22 0 8A 0 0 0 80 A8 12 1 AD 0 0 0 5C 41 4F 21 0 0 0 0 0 0 0 0
19:40:07.548 -> packetCfg now valid
19:40:07.548 -> packetCfg classAndIDmatch
19:40:07.582 -> waitForNoACKResponse: valid data with CLS/ID match after 976 msec
19:40:07.582 -> Lat: 444583798 Long: 261325910 (degrees * 10^-7) Alt: 116925 (mm) SIV: 20
```

I tried to understand every debugging message and it seems like there is no errors. 
Is 976 msec is a good transmission time?

Just another question, now that is it working, my goal is to do RTK. If I RTK working, does it will directly appears in my NMEA message the RTK status (since I have RTCM3 as protocol output)? Or do I will have to set up a new configuration?

PaulZC, 2024-05-24:
Hi Lucas (@Lucas0rt ),

Excellent - glad that's working for you.

This library uses UBX-CFG-PRT to establish if the GNSS module is connected. That will fail if the UBX input or output protocol is disabled. [v3 of the library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3) does that same thing but using the Configuration Interface VALGET message. Again, that will fail if the UBX input or output protocol is disabled.

You are _Polling_ the Position-Velocity-Time NAV-PVT message, so, yes, it can take up to one full navigation interval before the data is returned. The default is 1 second (1 Hz). You can make this more efficient by making the PVT message _Periodic_. In the library, we call this ["Auto"](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/Example13_PVT). You can also use [Callbacks](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/Callbacks) to help.

We have tutorials on how to set up an RTK base and rover:

* https://learn.sparkfun.com/tutorials/setting-up-a-rover-base-rtk-system

* https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station

And there are [code examples](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/ZED-F9P/Example3_StartRTCMBase/Example3_StartRTCMBase.ino) too.

For the Rover, the [PVT carrSoln (Carrier Solution)](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/f8047a525b45e627127a8215a54f6e381f836213/examples/ZED-F9P/Example17_NTRIPClient_With_GGA_Callback/Example17_NTRIPClient_With_GGA_Callback.ino#L126-L136) flag will tell you if it is in RTK Float or Fix.

Best wishes,
Paul

closed 2024-05-24 by PaulZC