sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #223: Ability to set Position Accuracy

cyclops1982, 2024-05-29:
### Can we set Position accuracy?
Hi, i'm wondering if it would be possible to set the Position Accuracy mask via UBX-CFG-NAV5 message in this library?

I can't find a method for this, and i'm wondering why it's not there.
Would a PR with that added be accepted?

### Your workbench
* RAK12500 (WisBlock GNSS GPS Location Module u-blox ZOE-M8Q)

PaulZC, 2024-05-30:
Hi Ruben (@cyclops1982 ),

If you want full control over NAV5, you could achieve that with a custom command - see [Example20](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example20_SendCustomCommand/Example20_SendCustomCommand.ino).

If you just want to change pAcc via posMask, then yes, sure, we would accept a PR. I'd suggest starting with ```setDynamicModel``` and adapt it for ```setPositionMask```. Add your new methods to .h, .c and keywords.txt.

Looking at the code for ```setDynamicModel```, I'm wondering if I should be using read-modify-write on the two mask bytes? I think [this line](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/f8047a525b45e627127a8215a54f6e381f836213/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L8220) should be deleted, and [this line](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/f8047a525b45e627127a8215a54f6e381f836213/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L8219) should actually be:

```
payloadCfg[0] |= 0x01; // Set the dyn bit - leave others unmodified
```

Please include those changes in your PR.

For posMask, you want ```payloadCfg[0] |= 0x10;``` and load your ```uint16_t pAcc``` into ```payloadCfg[18]``` and ```payloadCfg[19]``` - little endian.

Best wishes,
Paul

cyclops1982, 2024-05-30:
Thanks @PaulZC  - PR #224  created. Please have a look, very welcome to receive feedback!

PaulZC, 2024-05-30:
Resolved by #224

closed 2024-05-30 by PaulZC