sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #226: Increase delay between measurement

yriouvortex, 2024-07-17:
### Subject of the issue
Delay between measurement on the ZED-F9P is 65535 ms maximum. 
As data is read from UART and saved into ram buffer by calling checkUblox repeatedly there isn't any way to decrease the quantity of data that get logged into ram buffer.
I want to achieve a slower data collection by saving per example 1 measurement every 10 for long time positionning. Is there any way to get an interrupt info when new data is available ? 
I thought that callback could achieve that (at every call I would increase a counter and when it reaches a given number, I could save the data and reset the counter) but they are also polled in a while loop repeatedly. 

### Your workbench
Code is running on ESP32-S2 with Arduino Framework

### Steps to reproduce
Init uart to 9600
```
    // Software reset the module
        gnss.softwareResetGNSSOnly();

        // Set the UART1 to output RAW data
        gnss.setUART1Output(COM_TYPE_UBX);

        gnss.enableDebugging();

        gnss.setVal16(UBLOX_CFG_RATE_NAV, 0x01);

        // Lowest frequency possible
        gnss.setVal16(UBLOX_CFG_RATE_MEAS, 0x3e8);

        // Set time to be UTC
        gnss.setVal16(UBLOX_CFG_RATE_TIMEREF, 0);

        // Disable beidou satellites
        gnss.setVal16(UBLOX_CFG_SIGNAL_BDS_ENA, 0);

        // Disable sbas satellites
        gnss.setVal16(UBLOX_CFG_SIGNAL_SBAS_ENA, 0);

        // Enable automatic RXM RAWX messages: without callback; without implicit update
        gnss.setAutoRXMRAWX(true, false);

        // Enable RXM RAWX data logging
        gnss.logRXMRAWX();
        delay(100);
```

PaulZC, 2024-07-17:
Hi Yannick (@yriouvortex ),

If I understand you correctly, you want to log only RXM-RAWX every 10 seconds?

You can achieve that by setting CFG-RATE-MEAS to 10000: ```gnss.setVal16(UBLOX_CFG_RATE_MEAS, 10000);```

Or, set CFG-RATE-MEAS to 1000, and set CFG-RATE-NAV to 10: ```gnss.setVal16(UBLOX_CFG_RATE_NAV, 10);```

Or, set CFG-RATE-MEAS to 1000, set CFG-RATE-NAV to 1, and set the rate of the RAWX message to 10: replace ```gnss.setAutoRXMRAWX(true, false);``` with ```gnss.setAutoRXMRAWXrate(10, false);```.

If this solves your problem, please close this issue.

Best wishes,
Paul

yriouvortex, 2024-07-18:
Hi, thank you for the answers, I want to achieve a higher delay of rawx message (at max it is 65535 ms defined by  gnss.setVal16(UBLOX_CFG_RATE_MEAS, delay)). I want to have for example a RAWX message every 5 minutes but I can't manage to understand how to achieve this .

PaulZC, 2024-07-18:
Hi Yannick (@yriouvortex ),

OK. I understand.

Is post-processing an option for you? It would be easy to write some Python to discard the unwanted RAWX messages from the log file.

Or, hack the library, and add the static counter here:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/ce4629bbbcfbbde96dde1a236c618e0800430a70/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L4092-L4096

It would look something like:

```
        // Check if we need to copy the data into the file buffer
        if (packetUBXRXMRAWX->automaticFlags.flags.bits.addToFileBuffer)
        {
            static int counter = 0;
            if (counter == 0)
                storePacket(msg);
            counter++;
            if (counter == 300)
                counter = 0;
        }
```

Best wishes,
Paul

PaulZC, 2024-07-19:
Closing... Please reopen if you need more help with this.
Best wishes,
Paul

closed 2024-07-19 by PaulZC