sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #227: base station location through rtcm data

jasvinder1079, 2024-07-18:
hello @nseidle ,
Thanks for the amazing work, I am new to gnss domain.
My question is, Does rtcm data consist of base location info also or only corrections ,if yes can we extract it in rover and how ?

PaulZC, 2024-07-18:
Hello Jasvinder (@jasvinder1079 ),

The best place to ask questions like this is the SparkFun GNSS Forum: https://community.sparkfun.com/c/global-positioning-system-gps/96

The RTCM1005 and RTCM1006 messages both contain the base location - in ECEF X/Y/Z format. The RTCM message format is proprietary to RTCM, you need to purchase the specification to see how the messages are constructed. But you will find some examples in v3 of the SparkFun GNSS library:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/ZED-F9P/Example24_getLatestRTCM1005/Example24_getLatestRTCM1005.ino

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/main/examples/ZED-F9P/Example25_RTCM1005_Callback/Example25_RTCM1005_Callback.ino

Best wishes,
Paul

closed 2024-07-18 by PaulZC

jasvinder1079, 2024-07-18:
Thanks @PaulZC ,
In the provided example ,the API u provide for base location giving in ECEF(x,y,z) format ,Can we convert this into lat long or utm coordinate ?And how accurate this is.

PaulZC, 2024-07-18:
The best place to ask questions like this is the SparkFun GNSS Forum: https://community.sparkfun.com/c/global-positioning-system-gps/96

There are standard numerical methods to do the conversion, but it depends on which datum (ellipsoid) you are using. The numbers will be different for WGS84 compared to NAD83.