sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #229: SPI interface is not working on Tracker SOM

TanayRayav, 2024-08-23:
I am using SparkFun-u-blox library and interfacing with ublox NEO-M8U modem on Tracker SOM product.

Interface between Neo ublox-M8U and board is SPI.

Create a test sample and the device gets stuck in myGNSS.begin(SPI, UBLOX_CS_PIN, 4000000)

Here are the SPI configurations that I used based on the board settings
#define UBLOX_CS_PIN                      (GPS_CS)
#define UBLOX_PWR_EN_PIN                  (GPS_PWR)
#define UBLOX_RESETN_PIN                  (GPS_RST)
#define UBLOX_TX_READY_MCU_PIN            (GPS_INT)

I am using Particle workbench and deviceOS@6.1.1

This is the board - https://docs.particle.io/assets/pdfs/datasheets/tracker-som-eval-board.pdf

Attached is a sample file.

[TestSample.zip](https://github.com/user-attachments/files/16723971/TestSample.zip)

PaulZC, 2024-08-23:
Hi @TanayRayav ,

I do not have the Tracker SOM and I do not use Particle, so I will be able to give you very little help here.

Looking at your ```Test.cpp```:

You have defined ```UBLOX_PWR_EN_PIN``` and ```UBLOX_RESETN_PIN```, but you are not using them. I suggest pulling them both high. If the reset pin is low, the GNSS is held in reset and will not communicate. V_GPS is probably switched by the PWR_EN pin, but I do not know if it should be high or low. Try high first.

I am closing this as this is as much help as I can give you. If you need more help, I suggest asking on the Particle forums.

I hope this helps,
Paul

closed 2024-08-23 by PaulZC

TanayRayav, 2024-08-23:
Hi @PaulZC ,
Thanks for the assistance. I will try to implement the changes as you suggested.
Regards,
Tanay