sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #231: I can't understand how RTK works

Nocentinia, 2024-11-28:
hello guys,
sorry if I'm writing here but I don't know where to make other requests.
I'm working on a project and I need accurate gps accuracy (especially speed and heading).
I'm currently using m8Q and I don't know how to make it more accurate.
from what I understand, a GPS RTK would be needed, but I can't figure out how it works!
do I need to have an internet connection? do I need to connect somewhere? if so, how do I do it? from the example, it's not very clear, some documentation should be added!
I'll point out that my system will be at sea (near the coast).
thank you so much for the answers and sorry for writing here!

PaulZC, 2024-11-28:
Hi @Nocentinia ,

For RTK, you need an RTK-capable module. The u-blox MAX-M8Q is not RTK-capable... You need a module like the NEO-M8P (single band) or ZED-F9P (dual band). We also have RTK-capable modules from Septentrio (mosiac-X5), Unicore (UM980) and Quectel (LG290P). The RTK corrections can come from another GNSS (Base), via Internet (NTRIP or MQTT), or L-Band. Please have a look at the hook-up guides and tutorials for our RTK boards. 

Best wishes,
Paul

closed 2024-11-28 by PaulZC

Nocentinia, 2024-11-28:
hello,
thanks for your prompt reply.
so I should have an arduino that acts as a base and generates a server.
another arduino that connects via internet to this base to get the correction data right?
aren't there any ready-made servers to connect to?

PaulZC, 2024-11-28:
Please read the hook-up guides and tutorials...

If your RTK Rover is at sea, it needs corrections from a fixed RTK Base.
The corrections could come from a local base via a radio link.
If the RTK Rover can connect to the Internet, the corrections could be delivered that way. From an NTRIP Caster (RTK2go) or a correction service like PointPerfect. You could use Bluetooth to connect to a mobile phone, and have the mobile phone connect to the Caster.
Or you can use L-Band (satellite) corrections. u-blox PointPerfect has an L-Band service; you need a subscription plan and a separate receiver (NEO-D9S) to receive the corrections.

If you do not need to know the Rover position in real time, you can have the Rover log "raw" GNSS data (u-blox RXM-RAWX and RXM-SFRBX messages), and use that to post-process the Rover's position accurately.

I hope this helps.

Nocentinia, 2024-11-29:
Sorry my friend, but I really can't find any guides and tutorials in your guides. just some simple examples!

PaulZC, 2024-11-30:
Follow [this link](https://www.sparkfun.com/search/results?term=rtk) and select the "TUTORIALS" tab:

[image]