sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #233: base64_encode in example probably needs updating

curiores, 2025-04-02:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/4e542ba29be98088e5a1c4ee9ccafdd07901cb86/examples/ZED-F9P/Example15_NTRIPClient/Example15_NTRIPClient.ino#L163

Not sure if it's true for the arduino <base64.h> but at least for the ESP32 header that I'm using the function call

base64_encode

should be updated to

base64::encode

see

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/base64.h

PaulZC, 2025-04-03:
Hi @curiores ,

If you are using ESP32 then these lines apply:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/4e542ba29be98088e5a1c4ee9ccafdd07901cb86/examples/ZED-F9P/Example15_NTRIPClient/Example15_NTRIPClient.ino#L152-L155

And so it should compile OK. If you are seeing compiler errors, please paste them here and tell us which board package you are using and which board you have selected.

Best wishes,
Paul

curiores, 2025-04-03:
You're right, that would cover it. Thanks @PaulZC -- I didn't use the full example just snippets.

closed 2025-04-03 by curiores