sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #235: Unable to get lat and long data

itsghaniz, 2025-04-19:
I’m currently working with the ZED-F9P GNSS module connected to an Arduino Nano ESP32 controller via I2C, using the official Arduino library provided for the ZED-F9P. The I2C communication appears to be established correctly—I can successfully detect the module's I2C address and run the provided example sketches.

However, I’m encountering an issue where latitude and longitude data are not being received. For instance:

When running Example 1 from the ZED-F9P section, the output does not include valid positional data.

[image]

Similarly, in Example 10, although the sketch runs, the output fails to provide any GNSS fix or positional updates.

[image]

the following connections have been established

nano A4 ==> ZED F9P SDA
nano A5 ==> ZED F9P SCL

Could you please advise on what could be causing this issue? 

Thank you in advance for your guidance.

PaulZC, 2025-04-20:
Hi @itsghaniz ,

For the ZED-F9P, we recommend version 3 of this library:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3

But this library (v2) should still work OK.

Forgive me for asking the obvious questions:

* Have you connected an antenna?
* Has the antenna got a clear view of the sky?
* Which ZED-F9P board do you have? Is it a SparkFun board?
* Have you tried communicating with the board using u-center (on a Windows PC)? What do you see? Is the ZED receiving GNSS signals?

Best wishes,
Paul

itsghaniz, 2025-04-22:
Hi, Hope you are doing well, 
Yes, that was the problem, I wasn't using the proper antenna, and was trying to get the coordinates in an environment that did not have a clear sky. I got a better antenna and went to a clear sky, and now it is working. 

Thanks a lot for the suggestion.

closed 2025-04-22 by PaulZC