sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #236: Unable to use GALILEO with UBlox M8Q due to NMEA protocol version not being changable

ChSt15, 2025-04-30:
Im using a UBlox M8Q and want to enable Galileo. This I can enable it but the satellites are not used. This has the same behaviour in UCenter. To be able to use Galileo, you must hange the NMEA Protocol from 4.0 to 4.1. Then is works, but as far as i see in the Sparkfun Library, there is no way to change the NMEA Protocol to 4.1

PaulZC, 2025-04-30:
Hi @ChSt15 ,

Please clarify:
* Are you using the MAX-M8Q?
* Are you using only UART1 to communicate with the module?

Perhaps this is a baud rate issue? NAV-PVT is 100 bytes. At 10Hz, that is >10,000 bits/second. This will overload UART1 at the default of 9600 baud.

Please try increasing the baud rate, or decrease the navigation rate. What happens then?

Best wishes,
Paul

ChSt15, 2025-05-01:
Im using the SAM-M8Q in a breakout board.
Im only using the UART port. I have the UART port set to 115200 baud and can communicate through my flight controller with UCenter.
I found, if reset the module using UCenter and reconfigure it to use Galileo, then the Galileo works and I can see the Galileo satellites in Ucenter when using NMEA protocol 4.1. If I then disable NMEA over the UART port, and enable PVT and only Galileo sats, then i eventually get a position estimate using only Galileo. 
If i factory reset the module and then use the previously mentioned commands with the library to configure the module directly, then i never get satellites reported or a lock. Enabling GLONASS or GPS does work.

PaulZC, 2025-05-01:
Hi @ChSt15 ,

OK. I understand. But I don't really see how we can help. It sounds like the module and this library are working normally. All I can suggest is that you break the issue down, change one setting at a time, and find the point at which you no longer get solutions. Ensure Galileo is enabled _before_ you disable the other constellations. The M8 can only receive three constellations simultaneously, so you may need to disable e.g. GLONASS first, then enable Galileo, wait, then disable the other constellations. Disable SBAS and QZSS, then disable GPS. IMES is disabled by default.

I hope this helps,
Paul

closed 2025-05-01 by PaulZC

PaulZC, 2025-05-07:
Hi @ChSt15 ,

Apologies. I had forgotten how primitive `enableGNSS` is in this version of the library. Please see issue #237 for more details.

Best wishes,
Paul