sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #240: GPS data if no power

jan5650, 2025-08-20:
Hi! 

I am building a device that continuously monitors the number of satellites. 
As long as it works, there is no problem, but if the GPS module's power supply is interrupted, I always get the last value back for the satellites number.

The  function does not recognize that there is no data left and returns the last data to the satellites, even though there is no usable data left.

I am using this:

```
myGNSS.checkUblox();
satelites = myGNSS.getSIV();

if satelites<4 { do sometgin}
```

jan5650, 2025-08-20:
I'm using this code:

```
myGNSS.checkUblox();
if (myGNSS.getPVT())
  {satelites = myGNSS.getSIV();}

if satelites<4 { do something}
```

Sadly the program never run  "do something". The ARduino don't know that there is no GPS data, its got the last satelite number constant.

PaulZC, 2025-08-21:
If you are using "Auto" (periodic) PVT messages (`setAutoPVTrate`) then the `checkUblox` is actually redundant. `getPVT` will call `checkUbloxInternal` for you, behind the scenes.

Perhaps this would work?

```
bool newPVT = myGNSS.getPVT();
if (newPVT)
  satelites = myGNSS.getSIV();
if (newPVT && (satelites < 4))
{
  // do something
}
else
{
  // do something else
}
```